skywodd/pcf8574_arduino_library #7: not work with ESP8266

452, 2016-07-16:
```log
In file included from /home/ihor/Arduino/libraries/PCF8574/PCF8574.cpp:22:0:
/home/ihor/Arduino/libraries/PCF8574/PCint.h:3:41: error: 'PCMSK0' was not declared in this scope
 volatile uint8_t *port_to_pcmask[] = { &PCMSK0, &PCMSK1, &PCMSK2 };
                                         ^
/home/ihor/Arduino/libraries/PCF8574/PCint.h:3:50: error: 'PCMSK1' was not declared in this scope
 volatile uint8_t *port_to_pcmask[] = { &PCMSK0, &PCMSK1, &PCMSK2 };
                                                  ^
/home/ihor/Arduino/libraries/PCF8574/PCint.h:3:59: error: 'PCMSK2' was not declared in this scope
 volatile uint8_t *port_to_pcmask[] = { &PCMSK0, &PCMSK1, &PCMSK2 };
                                                           ^
/home/ihor/Arduino/libraries/PCF8574/PCint.h: In function 'void PCattachInterrupt(uint8_t, void (*)(), int)':
/home/ihor/Arduino/libraries/PCF8574/PCint.h:42:2: error: 'PCICR' was not declared in this scope
  PCICR |= 0x01 << port;
  ^
/home/ihor/Arduino/libraries/PCF8574/PCint.h: In function 'void PCdetachInterrupt(uint8_t)':
/home/ihor/Arduino/libraries/PCF8574/PCint.h:67:3: error: 'PCICR' was not declared in this scope
   PCICR &= ~(0x01 << port);
   ^
/home/ihor/Arduino/libraries/PCF8574/PCint.h: At global scope:
/home/ihor/Arduino/libraries/PCF8574/PCint.h:104:7: error: expected constructor, destructor, or type conversion before '(' token
 SIGNAL(PCINT0_vect) {
       ^
/home/ihor/Arduino/libraries/PCF8574/PCint.h:107:7: error: expected constructor, destructor, or type conversion before '(' token
 SIGNAL(PCINT1_vect) {
       ^
/home/ihor/Arduino/libraries/PCF8574/PCint.h:110:7: error: expected constructor, destructor, or type conversion before '(' token
 SIGNAL(PCINT2_vect) {
       ^
exit status 1
Error compiling for board ESPDuino (ESP-13 Module).
```

itsjustvenky, 2016-07-16:
This is only for Arduino and this awesome library. For esp8266 there is another simple library but not as this much.

452, 2016-07-16:
I use esp8266 with https://github.com/esp8266/Arduino
this library need to use Arduino APIs layer

itsjustvenky, 2016-07-16:
I don't like to post other library link here you can search esp8266 pcf8574

itsjustvenky, 2016-07-16:
By the way if you are planning to use interrupt process then it won't work well as on this library. Arduino + pcf works well compared to esp + pcf . This issue is only with interrupt and not GPIO

AndersV209, 2018-12-04:
what is needed to make this Library work with esp8266 ? i like this one and not the simpler version