skywodd/pcf8574_arduino_library #11: Arduino Nano crashing on digitalRead

saspol, 2018-02-18:
Hello,
Im trying to get expander pins state on interrupt receiving and always got crash.
Can you please see whats problem with this code?

`#include <Wire.h>
#include "PCF8575.h"

PCF8575 expander;

void setup() {
  Serial.begin(57600);
  expander.begin(0x20);
  expander.pinMode(0, INPUT);
  expander.enableInterrupt(2, ISRgateway);
}

void ISRgateway() {
  Serial.println("Board interrupt!");
  Serial.println(expander.read(), DEC);
}

void loop() {

}`

skywodd, 2018-02-19:
Hello,

Try replacing the Serial.println with a led blink or something simple to see if the isr is called.
Or set a boolean flag in the isr, and print from the loop() function.

This code is like ... 5 year old, I don't really remember how the interruption code work.
I still remember something about Wire (which is called by expander.read()) not liking at all to be called from an interruption function. Maybe this is still true today.

saspol, 2018-02-19:
Hi, i set flag in isr and then do read in loop. This working as expected.
Thanx!