skywodd/pcf8574_arduino_library #12: digitalRead returns wrong values ?

chopperaddict, 2018-05-27:
I have just started using tis library, and it is great so far writing to the chip, BUT I want to be able to read its content as well, but the following code returns 
1111010
1001

`   uint8_t input = 0;
   setStartupFrogs();
   Serial.println("Startup setting setup");
   delay(3000);
   //Serial.println(expander.digitalRead(3) ? "HIGH" : "LOW"); // Print button pin state
   input = Serial.println(expander.read(), BIN); // Read the whole pins input register
   Serial.println(input, BIN);
`
setStrartupFrogs() actually sets pins 0, 2,and 7, so I am expecting to see

10000101

assuming it maintains MSB or LSB forread and write, but these values are not correct either way around ?

Is this a known issue ?

FernandoGarcia, 2018-10-05:
I have reported this issue before.

https://github.com/skywodd/pcf8574_arduino_library/issues/3

francescolavra, 2018-10-26:
`Serial.println()` returns the number of characters written to the serial port, so when you execute `input = Serial.println(expander.read(), BIN)` you are assigning to the `input` variable the number of characters written, which equals the size of the string "1111010" plus the "\r\n" sequence, i.e. 9 characters; that's why you see 1001.

pierrot10, 2019-02-04:
@francescocolavra, Ok than could show use how we can  display in a terminal or a LCD when the pin is HIGH?

francescolavra, 2019-02-05:
Like that:
`Serial.println(expander.digitalRead(myPin) ? "HIGH" : "LOW");`

pierrot10, 2019-02-09:
Dera Francesco, thank for your reply