skywodd/pcf8574_arduino_library #15: can not get it working on wemos lolin esp32

sprokkie, 2018-11-27:
Hi when compiling getting this error:

\Arduino\libraries\PCF8575\PCint.h:3:41: error: 'PCMSK0' was not declared in this scope
 volatile uint8_t *port_to_pcmask[] = { &PCMSK0, &PCMSK1, &PCMSK2 };
                                         ^
\Arduino\libraries\PCF8575\PCint.h:3:50: error: 'PCMSK1' was not declared in this scope
 volatile uint8_t *port_to_pcmask[] = { &PCMSK0, &PCMSK1, &PCMSK2 };
                                                  ^
\Arduino\libraries\PCF8575\PCint.h:3:59: error: 'PCMSK2' was not declared in this scope
 volatile uint8_t *port_to_pcmask[] = { &PCMSK0, &PCMSK1, &PCMSK2 };
                                                           ^
Arduino\libraries\PCF8575\PCint.h: In function 'void PCattachInterrupt(uint8_t, void (*)(), int)':
\Arduino\libraries\PCF8575\PCint.h:42:2: error: 'PCICR' was not declared in this scope
  PCICR |= 0x01 << port;
  ^
\Arduino\libraries\PCF8575\PCint.h: In function 'void PCdetachInterrupt(uint8_t)':
D:\Ferry\Mijn documenten\Arduino\libraries\PCF8575\PCint.h:67:3: error: 'PCICR' was not declared in this scope
   PCICR &= ~(0x01 << port);
 ^
\Arduino\libraries\PCF8575\PCint.h: At global scope:
\Arduino\libraries\PCF8575\PCint.h:104:7: error: expected constructor, destructor, or type conversion before '(' token
 SIGNAL(PCINT0_vect) {
       ^
\Arduino\libraries\PCF8575\PCint.h:107:7: error: expected constructor, destructor, or type conversion before '(' token
 SIGNAL(PCINT1_vect) {
      ^
\Arduino\libraries\PCF8575\PCint.h:110:7: error: expected constructor, destructor, or type conversion before '(' token
 SIGNAL(PCINT2_vect) {
       ^

i added:
#include <Wire.h>    // Required for I2C communication
#include "PCF8575.h" // Required for PCF8575

/** PCF8575 instance */
PCF8575 expander; 

 /* Start I2C bus and PCF8575 instance */
  expander.begin(0x20);
  
  /* Setup some PCF8575 pins for demo */
  expander.pinMode(0, OUTPUT);
  expander.pinMode(1, OUTPUT);
  expander.pinMode(2, OUTPUT);
  expander.pinMode(3, OUTPUT); 

  server.on("/Fan/toggle", HTTP_GET, [](AsyncWebServerRequest *request){
  digitalWrite(Fan, !digitalRead(Fan));  
  expander.toggle(1);  <----- this line, the others are working in wemos pin's
  request->send(200, "text/plain", String(!digitalRead(Fan)));

dont know what is wrong

sprokkie, 2018-11-29:
found it out,
i commended the line's about interupt in the pcf8575.h file
now it is working

masteriii, 2019-05-29:
Not working on node mcu v3

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\masteriii\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\masteriii\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\masteriii\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=80,vt=flash,exception=disabled,ssl=all,eesz=4M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=57600 -vid-pid=1A86_7523 -ide-version=10809 -build-path C:\Users\MASTER~1\AppData\Local\Temp\arduino_build_630057 -warnings=none -build-cache C:\Users\MASTER~1\AppData\Local\Temp\arduino_cache_434506 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python.path=C:\Users\masteriii\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.python-3.7.2-post1.path=C:\Users\masteriii\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\masteriii\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=C:\Users\masteriii\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs.path=C:\Users\masteriii\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=C:\Users\masteriii\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -verbose C:\Users\MASTER~1\AppData\Local\Temp\arduino_modified_sketch_197218\ButtonBlink.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\masteriii\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\masteriii\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\masteriii\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=80,vt=flash,exception=disabled,ssl=all,eesz=4M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=57600 -vid-pid=1A86_7523 -ide-version=10809 -build-path C:\Users\MASTER~1\AppData\Local\Temp\arduino_build_630057 -warnings=none -build-cache C:\Users\MASTER~1\AppData\Local\Temp\arduino_cache_434506 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python.path=C:\Users\masteriii\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.python-3.7.2-post1.path=C:\Users\masteriii\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\masteriii\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=C:\Users\masteriii\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs.path=C:\Users\masteriii\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=C:\Users\masteriii\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -verbose C:\Users\MASTER~1\AppData\Local\Temp\arduino_modified_sketch_197218\ButtonBlink.ino
Using board 'nodemcuv2' from platform in folder: C:\Users\masteriii\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2
Using core 'esp8266' from platform in folder: C:\Users\masteriii\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2
Detecting libraries used...
"C:\\Users\\masteriii\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\masteriii\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/include" "-IC:\\Users\\masteriii\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/lwip2/include" "-IC:\\Users\\masteriii\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\MASTER~1\\AppData\\Local\\Temp\\arduino_build_630057/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10809 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DFLASHMODE_DIO -DESP8266 "-IC:\\Users\\masteriii\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\cores\\esp8266" "-IC:\\Users\\masteriii\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\variants\\nodemcu" "C:\\Users\\MASTER~1\\AppData\\Local\\Temp\\arduino_build_630057\\sketch\\ButtonBlink.ino.cpp" -o nul
"C:\\Users\\masteriii\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\masteriii\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/include" "-IC:\\ [... cut ...]
       ^

C:\Users\masteriii\Documents\Arduino\libraries\PCF8575\PCint.h:110:7: error: expected constructor, destructor, or type conversion before '(' token

 SIGNAL(PCINT2_vect) {

       ^

Using library Wire at version 1.0 in folder: C:\Users\masteriii\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Wire 
Using library PCF8575 in folder: C:\Users\masteriii\Documents\Arduino\libraries\PCF8575 (legacy)
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

khvalera, 2019-08-17:
I confirm the error when working with nodemcu :(